blue-systems/plamo-initial-dev_DEPRECATED #269: Discover can't be installed during phone image build

bhush9, 2017-12-12:
```
Investigating (0) plasma-phone [ armhf ] < none -> 34+p16.04+git20171113.1352 > ( metapackages )
Broken plasma-phone:armhf Depends on flatpak [ armhf ] < none -> 0.9.3-0alexlarsson2~xenial+16.04+xenial+build5 > ( admin )
  Considering flatpak:armhf 2 as a solution to plasma-phone:armhf 9999
  Re-Instated libarchive13:armhf
    Reinst Failed early because of libostree-1-1:armhf
Investigating (0) plasma-discover-flatpak-backend [ armhf ] < none -> 5.11.4+p16.04+git20171205.0155-0 > ( kde )
Broken plasma-discover-flatpak-backend:armhf Depends on flatpak [ armhf ] < none -> 0.9.3-0alexlarsson2~xenial+16.04+xenial+build5 > ( admin )
  Considering flatpak:armhf 2 as a solution to plasma-discover-flatpak-backend:armhf 0
  Holding Back plasma-discover-flatpak-backend:armhf rather than change flatpak:armhf
Investigating (1) plasma-phone [ armhf ] < none -> 34+p16.04+git20171113.1352 > ( metapackages )
Broken plasma-phone:armhf Depends on flatpak [ armhf ] < none -> 0.9.3-0alexlarsson2~xenial+16.04+xenial+build5 > ( admin )
  Considering flatpak:armhf 2 as a solution to plasma-phone:armhf 9999
Broken plasma-phone:armhf Depends on plasma-discover-flatpak-backend [ armhf ] < none -> 5.11.4+p16.04+git20171205.0155-0 > ( kde )
  Considering plasma-discover-flatpak-backend:armhf 0 as a solution to plasma-phone:armhf 9999
    Reinst Failed because of flatpak:armhf
Done
Package 'phonon4qt5-backend-vlc' is not installed, so not removed
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 plasma-phone : Depends: flatpak but it is not going to be installed
                Depends: plasma-discover-flatpak-backend but it is not going to be installed
```

star-buck, 2017-12-12:
Are you sure discover REQUIRES flatpak?
It was always discussed as being optional.

aleixpol, 2017-12-12:
Yes, it's optional, only need to hold `plasma-discover-flatpak-backend`. (although I don't think I understand the error message you are getting)

bhush9, 2017-12-12:
Ah, sorry I realized now that it's plasma-discover-flatpak-backend is what is at fault.. I will revert my change and install just plasma-discover.

Sorry for confusion, will include in last image I will build before going off today.

bhush9, 2017-12-12:
(neverthless, discover build needs to be fixed on neon CI anyway)

jriddell, 2017-12-12:
now compiled in neon CI

bhush9, 2017-12-13:
Discover is also installed, so marking as Ready for QA.

star-buck, 2017-12-13:
Can I flash new img to test?

bhush9, 2017-12-13:
Yes latest image have it installed.

closed 2017-12-13 by star-buck