blue-systems/plamo-initial-dev_DEPRECATED #270: fix 20 seconds delay between start sound and login screen coming up

star-buck, 2017-12-12:


bhush9, 2017-12-18:
For some reason, for first few seconds, kwin fails to initialize disaplay it seems from initial investigation, I will try to figure this out.

bhush9, 2017-12-19:
I investigated it more and I've at least solved the initial delay before starting android container locally, but kwin_wayland still takes time to render, I am not sure what is wrong about that yet, I will investigate more tomorrow

bhush9, 2018-01-29:
`20180124-113114` image have this fixed.

random note: instead of lockscreen, it will boot to plasmashell as lockscreen on startup is disabled, based on feedback in matrix and telegram lockscreen confuses people since they have not enabled it.. this haven't disabled lockscreen completely but just on startup..

closed 2018-02-08 by star-buck