blue-systems/plamo-initial-dev_DEPRECATED #275: Chromium: uses "old" keyboard, some buttons like enter, backspace, etc. dont work and crash chromium immediately

star-buck, 2017-12-12:


bhush9, 2017-12-12:
Not much we can do given the qtvirtualkeyboard offers keyboard only for Qt applications and not gtk applications..

bhush9, 2017-12-12:
IIRC there was plan to make qtvirtualkeyboard provide maliit interface to get this working, adding notmart to ticket

notmart, 2017-12-13:
> IIRC there was plan to make qtvirtualkeyboard provide maliit interface to get this working, adding notmart to ticket

not that i know, the problem as far i see is that chromium (and firefox) are xwayland, while the new keyboard looks like is wayland only.
if the problem is just the old keyboard layout we can adapt it, from the description of the problem seems that chromium is actually crashing?