blue-systems/plamo-initial-dev_DEPRECATED #277: vplayer: doesnt playback any video file (tried mp4, mkv, webm)

star-buck, 2017-12-13:
it displays stuck at 0:00/0:01, so i guess it's missinmg something for playback/decoding?

bhush9, 2017-12-13:
Yes I had plan to work on QtMultimedia bits today, but unexpectedly there was issue which prevented image from booting. So couldn't finish it.. 

I'll work on it tomorrow

star-buck, 2017-12-13:
here is whats needed:
"sudo apt install ubuntu-restricted-extras gstreamer1.0-plugins-bad gstreamer1.0-libav libavcodec-extra"

Works perfectly with HD video playback :)

star-buck, 2017-12-13:
so just installing those packages will make everything work!

bhush9, 2017-12-13:
That won't support all the video formats unfortunately, and also that video is software rendered so it's super laggy/and drops frames, I can install this package, but real solution is to get to gst-droid packaged.. which will also fix the camera

bhush9, 2017-12-13:
Also, gst-droid avoids issue of inclusion and distribution of non-free ubuntu-restricted extra in image..

star-buck, 2017-12-14:
I  am getting super smooth video playback, also support of most fileformats
I tried so far.

Please install as advised and spin a new img, also with audacioususoc
player.

bhush9, 2017-12-14:
IMG `20171214-165756` fixes this.

I however still prefer to get gst-droid included to avoid the licensing issues with ubuntu-restricted-extras.

closed 2017-12-15 by star-buck